hanruihua/neupan_ros #2: 缺少话题和node

987456tyy, 2025-04-07:
您好,我在运行了您代码中的roslaunch neupan_ros neupan_gazebo_limo.launch后出现警告:
TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint (parent odom) at time 9232.128000 according to authority unknown_publisher
此外,查看话题发现缺少/neupan_cmd_vel,/scan,/initial_path,/neupan_goal,等话题,rosnode list中也缺少neupan_control,
我使用anaconda3建立的环境python3.12运行roslaunch neupan_ros neupan_gazebo_limo.launch命令,
不知道是否是这导致的neupan_node没有运行

[image]

hanruihua, 2025-04-07:
你好

警告不影响，gazebo 正常启动了就可以。

NeuPAN control 报错 大概率是因为你的环境变量没有python，只有python3， 尝试安装` sudo apt-get install python-is-python3 `再运行。

987456tyy, 2025-04-07:
您好，我尝试安装了sudo apt-get install python-is-python3仍然没有改变，我的gazebo是正常运行的，而节点neupan_control却不正常，似乎没有运行neupan_node.run()

[image]
[image]

987456tyy, 2025-04-07:
> 你好
> 
> 警告不影响，gazebo 正常启动了就可以。
> 
> NeuPAN control 报错 大概率是因为你的环境变量没有python，只有python3， 尝试安装`sudo apt-get install python-is-python3`再运行。

您好，我尝试安装了sudo apt-get install python-is-python3仍然没有改变，我的gazebo是正常运行的，而节点neupan_control却不正常，似乎没有运行neupan_node.run()

[image]
[image]

hanruihua, 2025-04-07:
你好，neupan_node 看起来正常启动了，问题应该是在Gazebo里的模型没有正常加载。

- Gazebo 启动成功之后，障碍物应该会有随机运动，检查一下 rvo_ros的安装和变量设置。

- 目前的环境模型数量较多，如果电脑性能不够，容易出现模型崩溃的情况，可以尝试减少模型数量利用gazebo_world_complex_10.world 文件。

- 等Gazebo完全启动成功，模型全部正常加载运动之后再运行neupan control，否则limo模型容易崩。

PS: 尽量描述清楚不正常之处，小车是否有运动，neupan的neupan_cmd_vel 是否正常publish，Gazebo启动窗口的信息里在spawn model 阶段是否有报错，这样才好定位原因。

closed 2025-04-15 by 987456tyy