hanruihua/neupan_ros #4: no module named neupan

tdcsu, 2025-04-10:
Hi, thanks for the contribution of your great work!
After success compile and the initiation of the gazebo, when I attempt to run the following command:
`roslaunch neupan_ros neupan_gazebo_limo.launch` 
I met a error like `no module named neupan`，specifically:
`roslaunch neupan_ros neupan_gazebo_limo.launch
... logging to /home/td/.ros/log/75fbad88-15bf-11f0-8a96-7df56360907c/roslaunch-td-141733.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://td:41711/

SUMMARY
========

PARAMETERS
 * /neupan_control/base_frame: base_link
 * /neupan_control/config_file: /home/td/neupan_w...
 * /neupan_control/dune_checkpoint: /home/td/neupan_w...
 * /neupan_control/lidar_frame: laser_link
 * /neupan_control/map_frame: odom
 * /neupan_control/marker_size: 0.05
 * /neupan_control/scan_angle_range: -3.14 3.14
 * /neupan_control/scan_downsample: 6
 * /neupan_control/scan_range: 0.0 5.0
 * /rosdistro: noetic
 * /rosversion: 1.17.0

NODES
  /
    neupan_control (neupan_ros/neupan_node.py)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[neupan_control-1]: started with pid [141748]
process[rviz-2]: started with pid [141749]
Traceback (most recent call last):
  File "/home/td/neupan_ws/src/neupan_ros/src/neupan_node.py", line 3, in <module>
    from neupan_core import neupan_core
  File "/home/td/neupan_ws/src/neupan_ros/src/neupan_core.py", line 9, in <module>
    from neupan import neupan
ModuleNotFoundError: No module named 'neupan'
[neupan_control-1] process has died [pid 141748, exit code 1, cmd /home/td/neupan_ws/src/neupan_ros/src/neupan_node.py /scan:=/limo/scan /neupan_cmd_vel:=/cmd_vel /neupan_goal:=/move_base_simple/goal __name:=neupan_control __log:=/home/td/.ros/log/75fbad88-15bf-11f0-8a96-7df56360907c/neupan_control-1.log].
log file: /home/td/.ros/log/75fbad88-15bf-11f0-8a96-7df56360907c/neupan_control-1*.log
`
how can i sovle this issue?
thanks a lot !!!

hanruihua, 2025-04-10:
Hi，thanks for your attention.

This repo is the ROS wrapper of Neupan Planner. Please make sure you have installed the Neupan planner in your current Python by referring to https://github.com/hanruihua/NeuPAN: 

```
git clone https://github.com/hanruihua/NeuPAN
cd NeuPAN
pip install -e .  
```

tdcsu, 2025-04-10:
您好，我已收到您的邮件，会尽快处理~

tdcsu, 2025-04-10:
@hanruihua Thanks a lot, I do forget this important step, now this example works well with no error.

closed 2025-04-10 by tdcsu