hanruihua/neupan_ros #10: 您好，请问只使用点云如何实现自动驾驶？

wanghui-min, 2025-04-18:
目前我有一个点云地图、实时点云以及从初始位置规划好的路线，通过这些数据，如何实现自动导航避障呢？还是我还需要提供其他数据？

[image]
我看好像就需要三个数据吧？/neupan_goal话题是需要给啥啊？

我对ros不太懂，我是启动neupan_gazebo_limo.launch和run_limo_rviz.launch就可以把？然后发布LaserScan和Path数据，车需要初始化方向和path刚开始方向相同吗？不然是不是会方向不对？您可以介绍一下较为详细的流程吗？

hanruihua, 2025-04-18:
你好，

首先需要确保坐标系配置正确，可以检查/tf 的图，对应到launch 文件里的 "map_frame"， “base_frame”， 和 “lidar_frame” 即地图坐标系；机器人坐标系和lidar坐标系。

其次，将你的点云转为scan对应的topic，可以用 [pointcloud_to_laserscan](https://wiki.ros.org/pointcloud_to_laserscan) ros 包；初始位置规划好的路线直接对应发布到 /initial_path 就可以了。

/neupan_goal 的作用是从机器人当前位置和/neupan_goal的位置自行生成一条轨迹进行追踪，如果已经给定了/initial_path，可以不设置 /neupan_goal 

车的初始位置在 /initial_path 附近就可以了，算法会自动根据车的位置找到/initial_path  最近的点，最好和/initial_path 方向一致。否则会绕圈。

最后就是把neupan输出的control 命令对应到车的控制就可以了。只要topic的msg 类型一致，topic对应的名称用remap就可以，具体的命令都可以在ros wiki 上查询。

wanghui-min, 2025-04-21:
> 你好，
> 
> 首先需要确保坐标系配置正确，可以检查/tf 的图，对应到launch 文件里的 "map_frame"， “base_frame”， 和 “lidar_frame” 即地图坐标系；机器人坐标系和lidar坐标系。
> 
> 其次，将你的点云转为scan对应的topic，可以用 [pointcloud_to_laserscan](https://wiki.ros.org/pointcloud_to_laserscan) ros 包；初始位置规划好的路线直接对应发布到 /initial_path 就可以了。
> 
> /neupan_goal 的作用是从机器人当前位置和/neupan_goal的位置自行生成一条轨迹进行追踪，如果已经给定了/initial_path，可以不设置 /neupan_goal
> 
> 车的初始位置在 /initial_path 附近就可以了，算法会自动根据车的位置找到/initial_path 最近的点，最好和/initial_path 方向一致。否则会绕圈。
> 
> 最后就是把neupan输出的control 命令对应到车的控制就可以了。只要topic的msg 类型一致，topic对应的名称用remap就可以，具体的命令都可以在ros wiki 上查询。
> 
> 类似的架构可以参考rda_ros 里的自动驾驶的例子： https://github.com/hanruihua/rda_ros/tree/main/example/auto_driving

感谢您的回复，目前我是跑起来了，然后遇到了一些问题，想请教一下。
（1）点云我发布到scan，他并没有显示，当我发布到limo/scan的时候，发现可以显示并且其会有蓝色的路径指示拐弯等，不知道对不对。
（2）当我移动雷达的时候，发现小车并不会移动，是我必须自己每次进行实时定位发布odom数据吗？这个就是我初始化给发的数据
`import rospy
import tf

if __name__ == '__main__':
    rospy.init_node('tf_publisher')
    br = tf.TransformBroadcaster()
    rate = rospy.Rate(10.0)

    while not rospy.is_shutdown():
        # 发布地图坐标系 (odom) 到小车坐标系 (base_link) 的变换
        # 假设小车在地图中的位置是 (0, 0)，没有旋转平移
        br.sendTransform((0.0, 0.0, 0.0),
                         tf.transformations.quaternion_from_euler(0, 0, 0),
                         rospy.Time.now(),
                         "base_link",
                         "odom")

        br.sendTransform((0.0, 0.0, 0.0),
                         tf.transformations.quaternion_from_euler(0, 0, 0),
                         rospy.Time.now(),
                         "laser_link",
                         "base_link")
        rate.sleep()
`
还是这个算法会自己定位呢？或者是我的其他设置或者代码问题？

hanruihua, 2025-04-21:
你好，

1） 发布到limo/scan 有显示是因为 launch 文件里有remap的命令对 /scan的topic 映射到了新的名字，是针对gazebo的例子而设置的，你的情况如果想用scan 可以删除remap，能显示出来就是对的。

2） 如果你是手持雷达，那么需要跑一个定位算法，例如fast lio, lego slam, Cartographer.  这些算法都会发布一个坐标系的转换，从lidar 坐标系到 世界坐标系；neupan 算法会读取这个坐标系来定位机器人的位置和lidar点的坐标，然后发布控制命令。

3)  你的地图坐标系和小车坐标系相对固定了，小车肯定没法移动，应该是雷达和小车的坐标系固定，然后小车和世界的坐标系转换由定位算法发布。

4） 最简单的测试做法我建议是lidar 固定在小车上，然后只利用小车的odom发布的坐标关系来测试 （一般小车都会发布一个odom通过读取电机的转速)，这样不需要跑定位算法，测试没问题了，再配套定位算法来执行任务。

wanghui-min, 2025-04-23:
> 你好，
>
>     1. 发布到limo/scan 有显示是因为 launch 文件里有remap的命令对 /scan的topic 映射到了新的名字，是针对gazebo的例子而设置的，你的情况如果想用scan 可以删除remap，能显示出来就是对的。
>
>     2. 如果你是手持雷达，那么需要跑一个定位算法，例如fast lio, lego slam, Cartographer. [...]

非常感谢得到您细心的回复，目前我试验了neu和rda的代码，小车可以自己移动，但是还存在一点疑问
（1）发现neu设置允许初始化路径时提示没有该方法。

[image]
但是我看到neupan/blocks中是有这个py文件以及对应的set_initial_path方法。
（2）我使用rda_ros的时候发现设置所有都正常，然后小车移动确没有沿着预定路径行走，并且离起始点越走越远，目前还在测试是什么问题导致。使用neu_ros时，它会沿着默认路径进行行走，可能我的初始位置离路径很近，暂时没有发现其他问题。
（3）还有就是给其设置初始路径时，
 pose = PoseStamped()
pose.pose.orientation.w是否需要赋值？我看是关于旋转的，我没有赋值，按理说他根据点就连起来了路径。

wanghui-min, 2025-04-24:
> 你好
> 
> 1） 这个bug已经解决了， 你可以更新到最新的neupan版本 2） 建议小车的初始位置和和给定的轨迹初始位置一致，例如可以读取当前的机器人位置，以此为起点接入给定的轨迹。 3） 需要赋值，可以根据前后点之间的切线斜率赋值角度，因为path follow 的cost function 是会考虑到朝向的

感谢，目前已经可以自动驾驶了，但是有时候会提前提示已经到达目标点，但是当前位置和终点还有很远的距离，我通过数据模拟了一条行走的线路点和全局路径对其publish出去，发现偶尔也会有提前提示到达终点的现象，但是可以看到的是 小车还在绿色规划路径线上，并没有到终点，可能只走了三分之一的路程

wanghui-min, 2025-04-25:
> 你好
> 
> 1） 这个bug已经解决了， 你可以更新到最新的neupan版本 2） 建议小车的初始位置和和给定的轨迹初始位置一致，例如可以读取当前的机器人位置，以此为起点接入给定的轨迹。 3） 需要赋值，可以根据前后点之间的切线斜率赋值角度，因为path follow 的cost function 是会考虑到朝向的

我发现提前停止是因为两个地方所导致的，
`

    def check_curve_arrive(self, state, arrive_threshold=0.1, arrive_index_threshold=2):

        final_point = self.cur_curve[-1][0:2]
        arrive_distance = np.linalg.norm(state[0:2] - final_point)

        return (
            arrive_distance < arrive_threshold
            and self.point_index >= (len(self.cur_curve) - arrive_index_threshold)
        ) or (self.point_index >= (len(self.cur_curve) - 3))

`
是由于 or (self.point_index >= (len(self.cur_curve) - 3))而导致判断其提前到达了终点
第二个是由于

`

    def check_stop(self):
        return self.min_distance < self.collision_threshold

`
self.min_distance 我发现其运行了一段时间会小于0，所以导致认为发生了碰撞而导致停止，这个我看是模型输出的，是我给的变换矩阵出现了问题吗？但是我在rviz显示看一切都正常，如果有问题rviz显示也应该错误才对。

wanghui-min, 2025-04-25:
> > 你好
> > 1） 这个bug已经解决了， 你可以更新到最新的neupan版本 2） 建议小车的初始位置和和给定的轨迹初始位置一致，例如可以读取当前的机器人位置，以此为起点接入给定的轨迹。 3） 需要赋值，可以根据前后点之间的切线斜率赋值角度，因为path follow 的cost function 是会考虑到朝向的
> 
> 我发现提前停止是因为两个地方所导致的， `
> 
> ```
> def check_curve_arrive(self, state, arrive_threshold=0.1, arrive_index_threshold=2):
> 
>     final_point = self.cur_curve[-1][0:2]
>     arrive_distance = np.linalg.norm(state[0:2] - final_point)
> 
>     return (
>         arrive_distance < arrive_threshold
>         and self.point_index >= (len(self.cur_curve) - arrive_index_threshold)
>     ) or (self.point_index >= (len(self.cur_curve) - 3))
> ```
> 
> ` 是由于 or (self.point_index >= (len(self.cur_curve) - 3))而导致判断其提前到达了终点 第二个是由于
> 
> `
> 
> ```
> def check_stop(self):
>     return self.min_distance < self.collision_threshold
> ```
> 
> ` self.min_distance 我发现其运行了一段时间会小于0，所以导致认为发生了碰撞而导致停止，可能是我碰撞阈值和小车大小设置太大而导致的。

我发现问题所在了，是因为我给的路径点太少，我只给了路径中的起始点，转弯点和结束点所以存在以上问题，如果间隔10cm给他的话，他会按照路径去走，但是避障目前可能有问题，一旦距离到了阈值就自动停止，可能是我修改了部分源码导致的，我改回去再去测试一下，可能rda_ros的问题也是由于这个

wanghui-min, 2025-04-25:
> > > 你好
> > > 1） 这个bug已经解决了， 你可以更新到最新的neupan版本 2） 建议小车的初始位置和和给定的轨迹初始位置一致，例如可以读取当前的机器人位置，以此为起点接入给定的轨迹。 3） 需要赋值，可以根据前后点之间的切线斜率赋值角度，因为path follow 的cost function 是会考虑到朝向的
> > 
> > 
> > 我发现提前停止是因为两个地方所导致的， `
> > ```
> > def check_curve_arrive(self, state, arrive_threshold=0.1, arrive_index_threshold=2):
> > 
> >     final_point = self.cur_curve[-1][0:2]
> >     arrive_distance = np.linalg.norm(state[0:2] - final_point)
> > 
> >     return (
> >         arrive_distance < arrive_threshold
> >         and self.point_index >= (len(self.cur_curve) - arrive_index_threshold)
> >     ) or (self.point_index >= (len(self.cur_curve) - 3))
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ` 是由于 or (self.point_index >= (len(self.cur_curve) - 3))而导致判断其提前到达了终点 第二个是由于
> > `
> > ```
> > def check_stop(self):
> >     return self.min_distance < self.collision_threshold
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ` self.min_distance 我发现其运行了一段时间会小于0，所以导致认为发生了碰撞而导致停止，可能是我碰撞阈值和小车大小设置太大而导致的。
> 
> 我发现问题所在了，是因为我给的路径点太少，我只给了路径中的起始点，转弯点和结束点所以存在以上问题，如果间隔10cm给他的话，他会按照路径去走，但是避障目前可能有问题，一旦距离到了阈值就自动停止。
> 
> interval参数我不知道有没有作用？当我给他路径中的起始点，转弯点和结束点，他好像斜着直接去终点，如果我自己切分好10cm的给他，它会按照路径去走，但就是无法越障，会自动停止。

我给设置了一下min_radius，好像实现了避障，不知道是不是这个问题导致的，但是用的差速机器人，应该可以为0，也有可能是上次跑部分参数错误与min_radius没有关系

hanruihua, 2025-04-25:
你好，

1. 如果你给的路径点比较少，可以调小 interval的值，代表路径里的点的距离间隔，更小的interval 会生成更加密集的waypoints （前提是这个轨迹是通过neupan内部的`cg.generate_curve`函数生成的) . 

2. “当我给他路径中的起始点，转弯点和结束点，他好像斜着直接去终点，” 如果是通过neupan内部函数生成的curve, 你可以调小`interval `的值和 `ind_range`的值。
 
3. 目前到达终点的判定，一个是看index 是否到达阈值 （`arrive_index_threshold`），一个是根据距离判断（`arrive_threshold`). 

4. 出现 `self.min_distance` 小于0的情况的话，如果和现实的点的距离对不上，先检查一下你的DUNE model 是否是根据你的机器人尺寸训练的，目前默认提供的只针对limo小车的尺寸，根据你的小车，需要重新训练DUNE，并且在yaml里设置好。  其次，检查是否有什么噪点影响的。最后检查一下坐标系的转换是否正确，lidar和小车的相对位置尺寸设置是否正确。 

5. 对于差速轮，min_radius 应该没什么用影响。

wanghui-min, 2025-04-28:
> 你好，
> 
>     1. 如果你给的路径点比较少，可以调小 interval的值，代表路径里的点的距离间隔，更小的interval 会生成更加密集的waypoints （前提是这个轨迹是通过neupan内部的`cg.generate_curve`函数生成的) .
> 
>     2. “当我给他路径中的起始点，转弯点和结束点，他好像斜着直接去终点，” 如果是通过neupan内部函数生成的curve, 你可以调小`interval `的值和 `ind_range`的值。
> 
>     3. 目前到达终点的判定，一个是看index 是否到达阈值 （`arrive_index_threshold`），一个是根据距离判断（`arrive_threshold`).
> 
>     4. 出现 `self.min_distance` 小于0的情况的话，如果和现实的点的距离对不上，先检查一下你的DUNE model 是否是根据你的机器人尺寸训练的，目前默认提供的只针对limo小车的尺寸，根据你的小车，需要重新训练DUNE，并且在yaml里设置好。  其次，检查是否有什么噪点影响的。最后检查一下坐标系的转换是否正确，lidar和小车的相对位置尺寸设置是否正确。
> 
>     5. 对于差速轮，min_radius 应该没什么用影响。
> 
>     6. 你可以先按照你的尺寸参数在ir-sim里模拟运行一下，看看有没有问题，调好对应的adjust parameter,  再在实际小车上测试会比较顺利。

目前测试感觉没有什么问题了，唯一存在的问题就是那个路径划分的问题，如果只给几个拐弯点，就会存在问题，如果我自己切分publish就没有问题，不知道是因为切分路径后没有初始化最新路径还是怎么回事，但问题已经不大了，目前实验可以自己运动、避障等。
最后，我只能说一个字：牛！！！！！！！！！！！！！！！！！！！！！！！

hanruihua, 2025-05-07:
修复了set_initial_path 功能中存在的bug，能够更好的追踪轨迹，可以更新到最新的neupan 和 neupan_ros 版本查看效果

wanghui-min, 2025-05-08:
> 修复了set_initial_path 功能中存在的bug，能够更好的追踪轨迹，可以更新到最新的neupan 和 neupan_ros 版本查看效果

好的，目前我是没有发现什么问题了，室外用的rtk定位和定向，也实现了小车避障、自动驾驶等功能