hanruihua/neupan_ros #9: 请问一下rviz中无法运动小车是什么原因

tuanzi-zz, 2025-04-18:
您好,我在运行了您的代码后，在rviz中无法运动小车，查看节点后也没有查找到问题，想请问下这个是什么原因

[image]

[image]

[image]

hanruihua, 2025-04-18:
你好，

节点里没有neupan_control 说明没有启动成功，检查一下 neupan_control的节点和报错； 

也检查一下 limo模型是否崩溃。

neupan_control 最好在gazebo的环境已经彻底启动之后再运行 相关的launch命令

tuanzi-zz, 2025-04-18:
好的，谢谢您

戴洁
***@***.***

maenhui123, 2025-04-21:
您好，我也遇到了差不多的问题，好像neupan_control节点没启动起来，请问解决了吗

tuanzi-zz, 2025-04-22:
> 你好，
> 
> 节点里没有neupan_control 说明没有启动成功，检查一下 neupan_control的节点和报错；
> 
> 也检查一下 limo模型是否崩溃。
> 
> neupan_control 最好在gazebo的环境已经彻底启动之后再运行 相关的launch命令

您好，查看了neupan_control的节点，但是cmd_vel并没有发布，因此查不到相关日志，请问neupan算法的包需要怎么在ros中正常运行

tuanzi-zz, 2025-04-22:
您好，
查看了相关节点，发现cmd_vel并没有发布，因此没有相关的日志。然后rviz中无法运动也没有相关的轨迹线，猜测是neupan算法没有正常调用，但是已在bashrc中添加了neupan路径，并且在vsocde中查看neupan_node.py也可以找到定义的neupan.py，同时在roslaunch neupan_ros neupan_gazebo_limo.launch中也没有看到找不到neupan模块的报错，想和您请教下这个neupan包调用具体还需要什么步骤吗

[image]

> 你好，
> 
> 节点里没有neupan_control 说明没有启动成功，检查一下 neupan_control的节点和报错；
> 
> 也检查一下 limo模型是否崩溃。
> 
> neupan_control 最好在gazebo的环境已经彻底启动之后再运行 相关的launch命令

hanruihua, 2025-04-23:
你好，大概率是conda的bug导致的。

我更新了一版python3.8版本的 neupan: https://github.com/hanruihua/NeuPAN/tree/py38.

你可以选择安装这一版本，然后用系统默认的python3.8 运行所有的模块。

tuanzi-zz, 2025-04-25:
> 你好，大概率是conda的bug导致的。
> 
> 我更新了一版python3.8版本的 neupan: https://github.com/hanruihua/NeuPAN/tree/py38.
> 
> 你可以选择安装这一版本，然后用系统默认的python3.8 运行所有的模块。

好的，非常感谢大佬

closed 2025-04-25 by tuanzi-zz