hanruihua/neupan_ros #11: about python environment in ros

maenhui123, 2025-04-22:
您好，请问python环境怎么配置，我用的noetic版本，系统自带的是python3.8，但是安装neupan需要版本>=3.10，于是我又安装了python3.11，安装neupan相关依赖之后，在ros下仿真由于python版本的问题导致roslaunch卡住不能运行了，请问有什么好的解决办法吗？

[image]

hanruihua, 2025-04-23:
你好 你可以用conda建立两个虚拟环境python3.8和python3.10，python3.8 跑ros 部分，python3.10 跑neupan 部分。

不过针对这一问题，我更新了可以兼容python3.8 运行neupan的分支版本，你也可以选择安装这一版本来运行https://github.com/hanruihua/NeuPAN/tree/py38.    就可以利用系统的python3.8运行所有的模块了。

maenhui123, 2025-04-24:
您好，感谢您更新的python3.8版本，现在节点已经能启动了，在rviz中设置2D导航目标之后，小车正常出发，但是并没有触发避障，也没有朝着目标前进，查看下发的速度指令，发现都是沿x轴的，大小为0.1的线速度指令，请教一下您是什么原因，应该怎么解决？

[image]

[image]

[image]

hanruihua, 2025-04-25:
看着像是neupan中途报错了，得看看看节点的错误信息。

先更新下neupan到最新的版本，最近也修了几个bug 可能会导致问题。

closed 2025-05-12 by maenhui123