hanruihua/neupan_ros #14: gazebo仿真时遇到的一些问题

maenhui123, 2025-05-12:
您好，我将NeuPAN和neupan_ros都更新到了您发布的最新版本，但是我在运行example时，在rviz中启动后直接生成了初始轨迹，如下图，请问本来就应该是这样吗？怎么关闭呢？

[image]
还有就是小车并没有移动，查看neupan发布的/cmd_vel话题发现是下发了控制指令的，是gazebo的问题吗？

[image]

[image]

用2d nav goal设置了目标位姿之后好像也有问题，neupan规划的轨迹并没有更新

[image]

hanruihua, 2025-05-13:
你好  

gazebo的例子里会自动生成轨迹并且机器人会在A点和B点巡航。能正常publish命令说明neupan没有问题，但是速度 0.8太大了，limo模型容易崩，可以检查一下gazebo里的模型。

maenhui123, 2025-05-16:
您好，确实是模型崩了，请问在哪里修改参数呢，还有我打算在机器人上部署算法，但是输入目标点之后，neupan规划的轨迹是跟生成的初始路线对不上，查看cmd_vel话题也没有消息发布，请问是什么原因

[image]

[image]

[image]

[image]

hanruihua, 2025-05-16:
建议修改 yaml 里的 `ref_speed` 在 0.3 以下，否则limo的模型就会崩。类似的，也可以限制 `max_speed` 参数。

从rviz的显示来看，ros tf 有问题，检查一下  map_frame， base_frame, lidar_frame,  需要和 tf_tree 里的关系对应，尤其是map_frame的参数

maenhui123, 2025-05-16:
您好，不显示的问题已经解决了，但是我设置不同的goal，neupan规划的结果好像没变，下发的速度指令也可以证明这一点，我希望实现在rviz中设置一个goal，机器人能够规划出合理的轨迹到达设置的goal，请问还需要修改哪些地方？谢谢

[image]

[image]

hanruihua, 2025-05-17:
提供的limo gazebo的例子 可以通过 rviz的 2d nav_goal 给机器人设置新的goal，不需要修改。

maenhui123, 2025-05-19:
> 提供的limo gazebo的例子 可以通过 rviz的 2d nav goal 按钮给机器人设置新的goal，不需要修改。

是的，就是通过rviz设置的goal，但是规划的结果跟我给的goal完全无关，我是通过回放rosbag来初步验证规划结果，请问可能跟这个有关系吗

hanruihua, 2025-05-19:
是的  不能通过rosbag回放的方式跑规划，因为neupan ros 需要读取当前的tf,  需要按照给的例子，实时跑gazebo。

maenhui123, 2025-05-26:
您好，在虚拟环境中使用python3.11环境重新跑main分支，ir-sim和gazebo仿真都正常了，规划效果很好，可能是py3.8分支的问题，不排除本地系统的python3.8环境存在问题，最后感谢您耐心的解答以及开源如此优秀的工作！

hanruihua, 2025-05-26:
我测试了py38分支  没发现什么问题。  你可以提供下python版本和可复现的 ir-sim配置文件  我可以测试下是否是py38 分支存在bug。

maenhui123, 2025-06-04:
python版本是3.8.10
https://github.com/user-attachments/assets/a3052864-283f-4cf5-b936-50c559fb6885

[image]