hanruihua/neupan_ros #16: 在Rviz中遇到的问题

olmman, 2025-05-19:
您好，我希望在Gazebo中运行LIMO的示例，并且按照gazebo_limo文件夹下的readme.md中提到的sh gazebo_example_setup.sh命令安装了环境依赖。但是在启用Neupan Controer时，Rviz界面中没有任何显示，且RobotModel与LaserScan均存在报错，如图所示：

[image]

但Gazebo中能够正常加载机器人模型

[image]

我在工作空间下使用catkin_make重新编译后仍然存在该问题，以下是编译输出：
nubot@nubot:~/neupan_ws$ catkin_make
Base path: /home/nubot/neupan_ws
Source space: /home/nubot/neupan_ws/src
Build space: /home/nubot/neupan_ws/build
Devel space: /home/nubot/neupan_ws/devel
Install space: /home/nubot/neupan_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/nubot/neupan_ws/build"
####
####
#### Running command: "make -j32 -l32" in "/home/nubot/neupan_ws/build"
####
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_nodejs
[ 17%] Built target rvolib
[ 17%] Built target std_msgs_generate_messages_lisp
[ 17%] Built target geometry_msgs_generate_messages_eus
[ 17%] Built target geometry_msgs_generate_messages_cpp
[ 17%] Built target geometry_msgs_generate_messages_py
[ 23%] Built target gazebo_ros_ackerman
[ 29%] Built target multi_obs
[ 29%] Built target _limo_base_generate_messages_check_deps_LimoStatus
[ 29%] Built target _rvo_ros_generate_messages_check_deps_SetGoals
[ 35%] Built target limo_base_generate_messages_eus
[ 38%] Built target limo_base_generate_messages_cpp
[ 44%] Built target limo_base_generate_messages_py
[ 47%] Built target limo_base_generate_messages_nodejs
[ 50%] Built target limo_base_generate_messages_lisp
[ 52%] Built target rvo_ros_generate_messages_nodejs
[ 58%] Built target rvo_ros_generate_messages_py
[ 61%] Built target rvo_ros_generate_messages_lisp
[ 67%] Built target rvo_ros_generate_messages_eus
[ 70%] Built target rvo_ros_generate_messages_cpp
[ 70%] Built target limo_base_generate_messages
[ 82%] Built target limo_base_node
[ 82%] Built target rvo_ros_generate_messages
[ 88%] Built target rvo_node
[ 94%] Built target formation
[100%] Built target set_goals_client

hanruihua, 2025-05-20:
urdf 文件的tf 没有被正确的读取和发布，需要检查和测试一下  limo_ros 仓库 https://github.com/hanruihua/limo_ros，和在load模型的时候有关urdf的报错。

olmman, 2025-05-20:
> urdf 文件的tf 没有被正确的读取和发布，需要检查和测试一下 limo_ros 仓库 [https://github.com/hanruihua/limo_ros](https://bgithub.xyz/hanruihua/limo_ros) 和在load模型的时候有关urdf的报错。

olmman, 2025-05-20:
> 抱歉，刚刚由于误操作，将该问题关闭了，实际上我的问题并没有得到完全解决。 我在测试limo_ros仓库时，遇到了Resource not found: ydlidar_ros的问题，通过安装YDLidar-SDK 以及 ydlidar_ros_driver已经将该问题解决，但在运行roslaunch limo_bringup limo_teletop_keyboard.launch时提示 ERROR: cannot launch node of type [teleop_twist_keyboard/teleop_twist_keyboard.py]: teleop_twist_keyboard ROS path [0]=/opt/ros/noetic/share/ros ROS path [1]=/home/nubot/neupan_ws/src ROS path [2]=/opt/ros/noetic/share No processes to monitor shutting down processing monitor... ... shutting down processing monitor complete 我不清楚这是否会影响我在neupan_rosz中的使用。 而后我重新运行roslaunch neupan_ros neupan_gazebo_limo.launch，发现问题仍未得到解决，我查看了一下ros日志，发现存在该提示[roslaunch][ERROR] 2025-05-20 16:37:29,555: [neupan_control-2] process has died [pid 53505, exit code 127, cmd /home/nubot/neupan_ws/src/neupan_ros/src/neupan_node.py，似乎是neupan_node.py并未正确运行，而在查看该脚本后发现该脚本需要import neupan，这是否意味着我需要在neupan_ros工作目录下下载仓库https://github.com/hanruihua/NeuPAN并进行编译？如果需要，能否给出目录关系与安装顺序？

我在配置了NeuPAN的conda虚拟环境中再次运行gazebo示例，仍然存在问题，我检查运行roslaunch neupan_ros gazebo_limo_env_complex_20.launch的输出日志，发现controller_manager和robot_state_publisher都未正常启动
process[spawn_limo_model-4]: started with pid [63650]
ERROR: cannot launch node of type [controller_manager/spawner]: controller_manager
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/home/nubot/neupan_ws/src
ROS path [2]=/opt/ros/noetic/share
process[rvo_ros-6]: started with pid [63655]
ERROR: cannot launch node of type [robot_state_publisher/robot_state_publisher]: robot_state_publisher
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/home/nubot/neupan_ws/src
ROS path [2]=/opt/ros/noetic/share

olmman, 2025-05-20:
> 我在配置了NeuPAN的conda虚拟环境中再次运行gazebo示例，仍然存在问题，我检查运行roslaunch neupan_ros gazebo_limo_env_complex_20.launch的输出日志，发现controller_manager和robot_state_publisher都未正常启动 process[spawn_limo_model-4]: started with [...]

安装完robot_state_publisher和controller_manager后，rivz中已经能正常显示，且小车模型能够正常避障