hanruihua/neupan_ros #30: ros2执行出错wrong curve type Error in NeuPAN node: 'opt_state_list'

dengxingyu, 2025-06-30:
[我在ros2中尝试使用了另一个移植到ros2的项目，](https://github.com/huapu-kaf/NeuPAN-ROS2-Real-Robot)
但是运行过程中指定目标点后出现报错
# ros2 launch neupan_ros neupan_ros2.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-06-30-03-54-32-421936-dxy-Legion-Y9000P-IRX9-24380
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [neupan_node_ros2-1]: process started with pid [24381]
[neupan_node_ros2-1] (CVXPY) Jun 30 03:54:33 AM: Encountered unexpected exception importing solver DIFFCP:
[neupan_node_ros2-1] ImportError('diffcp >= 1.0.15 is required')
[neupan_node_ros2-1] [INFO] [1751255674.210128376] [neupan_node]: NeuPAN node initialized
[neupan_node_ros2-1] [INFO] [1751255674.383037151] [neupan_node]: Waiting for tf transform from base_link to map: "map" passed to lookupTransform argument target_frame does not exist. 
[neupan_node_ros2-1] [INFO] [1751255675.398507966] [neupan_node]: Waiting for tf transform from base_link to map: "map" passed to lookupTransform argument target_frame does not exist. 
[neupan_node_ros2-1] [INFO] [1751255676.414021590] [neupan_node]: Waiting for tf transform from base_link to map: "map" passed to lookupTransform argument target_frame does not exist. 
[neupan_node_ros2-1] [INFO] [1751255677.429518959] [neupan_node]: Waiting for tf transform from base_link to map: Lookup would require extrapolation into the past.  Requested time 3147.831000 but the earliest data is at time 3148.194000, when looking up transform from frame [base_link] to frame [map]
[neupan_node_ros2-1] [INFO] [1751255678.446004345] [neupan_node]: Waiting for tf transform from base_link to map: Lookup would require extrapolation into the past.  Requested time 3147.933000 but the earliest data is at time 3148.194000, when looking up transform from frame [base_link] to frame [map]
[neupan_node_ros2-1] [WARN] [1751255678.852691482] [neupan_node]: Waiting for neupan initial path
[neupan_node_ros2-1] [INFO] [1751255678.853828471] [neupan_node]: Scan obstacle points received
[neupan_node_ros2-1] [WARN] [1751255679.882212748] [neupan_node]: Waiting for neupan initial path
[neupan_node_ros2-1] [WARN] [1751255680.882801338] [neupan_node]: Waiting for neupan initial path
[neupan_node_ros2-1] [WARN] [1751255681.982267324] [neupan_node]: Waiting for neupan initial path
[neupan_node_ros2-1] [INFO] [1751255682.851748468] [neupan_node]: Set neupan goal: [-2.07, -0.02, -1.78]
[neupan_node_ros2-1] [INFO] [1751255682.851925720] [neupan_node]: Reference path update
[neupan_node_ros2-1] [INFO] [1751255682.882497796] [neupan_node]: Arrive at the target
[neupan_node_ros2-1] wrong curve type
[neupan_node_ros2-1] wrong curve type
[neupan_node_ros2-1] Error in NeuPAN node: 'opt_state_list'
[INFO] [neupan_node_ros2-1]: process has finished cleanly [pid 24381]

请问下应该从那方面分析错误原因

hanruihua, 2025-07-06:
你好

从报错信息上来看，大概率是因为  yaml 文件里没有配置好 ipath 相关的参数。可以参考 [neupan_planner_limo.yaml](https://github.com/huapu-kaf/NeuPAN-ROS2-Real-Robot/blob/main/neupan_ros-main/example/gazebo_limo/config/neupan_planner_limo.yaml#L18).

dengxingyu, 2025-07-07:
> 你好
> 
> 从报错信息上来看，大概率是因为 yaml 文件里没有配置好 ipath 相关的参数。可以参考 [neupan_planner_limo.yaml](https://github.com/huapu-kaf/NeuPAN-ROS2-Real-Robot/blob/main/neupan_ros-main/example/gazebo_limo/config/neupan_planner_limo.yaml#L18).

通过修改配置信息，和修改上面提到的ros2工程里面的一些数据类型转换问题（需要将float32转成float）,程序已经可以在ROS2中运行，但目前导航参数调整效果还不是很好，并且在发布一个目标点并到达之后，再发布下一个目标点时终端会一直输出已经到达，不会前往新的目标点。