hanruihua/neupan_ros #33: 请问作者,我该如何添加全局规划器呢,有时候需要绕过障碍物,如果生成的全局路径是直线就过不去

PestoQing, 2025-07-25:
请问下该如何添加呢,请问作者,我该如何添加全局规划器呢,有时候需要绕过障碍物,如果生成的全局路径是直线就过不去

hanruihua, 2025-07-27:
可以通过实时更新 initial path 来实现和 global planner的结合。

 neupan ros 节点会订阅 `/initial_path`（nav_msgs/Path）来读取 initial path, 因此 可以将其他全局规划器的输出， 例如 A star path，直接remap 到  `/initial_path`即可 (注意类型要相同)，同时将 `refresh_initial_path` 设置为 `True`，这样可以实时的读取全局规划器的path来不断更新initial path 。

PestoQing, 2025-07-27:
neupan_gazebo_limo.launch的代码
<launch> 
    <arg name="config_file" default='$(find neupan_ros)/example/gazebo_limo/config/neupan_planner_limo.yaml'/>
    <arg name="map_frame" default='map'/>
    <arg name="base_frame" default='base_link'/>
    <arg name="lidar_frame" default='laser_link'/>
    <arg name="scan_range" default='0.0 5.0'/>
    <arg name="scan_angle_range" default='-3.14 3.14'/>
    <arg name="marker_size" default='0.05'/>
    <arg name="marker_z" default='0.3'/>
    <arg name="scan_downsample" default='6'/>
    <arg name="dune_checkpoint" default='$(find neupan_ros)/example/gazebo_limo/pretrain_limo/model_5000.pth'/>

    
    <node pkg="map_server" type="map_server" name="map_server" args="$(find wpr_simulation)/maps/map.yaml"/>

    
    <node pkg="amcl" type="amcl" name="amcl">
        <param name="odom_frame_id" value="odom"/>
        <param name="base_frame_id" value="base_link"/>
        <param name="global_frame_id" value="map"/>
        <param name="initial_pose_x" value="0.0"/>
        <param name="initial_pose_y" value="0.0"/>
        <param name="initial_pose_a" value="0.0"/>
        <param name="update_min_d" value="0.1"/>
        <param name="update_min_a" value="0.1"/>
        <param name="min_particles" value="500"/>
        <param name="max_particles" value="2000"/>
        <param name="transform_tolerance" value="1.0"/>
        <remap from="scan" to="/limo/scan"/>
    </node>

    
    <node pkg="tf" type="static_transform_publisher" name="map_to_odom" 
          args="0 0 0 0 0 0 map odom 100"/>
    <node pkg="tf" type="static_transform_publisher" name="base_to_laser" 
          args="0.1 0 0.2 0 0 0 base_link laser_link 100"/>

    
    <node pkg="move_base" type="move_base" name="move_base" output="screen">
        <param name="base_global_planner" value="global_planner/GlobalPlanner" /> 
        <param name="base_local_planner" value="base_local_planner/TrajectoryPlannerROS"/>
        
        
        <param name="GlobalPlanner/allow_unknown" value="true"/>
        <param name="GlobalPlanner/use_dijkstra" value="true"/>
        <param name="GlobalPlanner/visualize_potential" value="true"/>
        
        <param name="GlobalPlanner/lethal_cost" value="253"/>
        <param name="GlobalPlanner/neutral_cost" value="50"/>
        <param name="GlobalPlanner/cost_factor" value="0.55"/>
        
        
        <remap from="/cmd_vel" to="/move_base_cmd_vel_unused"/>
        
        
        <remap from="/move_base/GlobalPlanner/plan" to="/initial_path"/>
        
        
        <remap from="/move_base/TrajectoryPlannerROS/global_plan" to="/initial_path"/>

        
        
        
        
        <rosparam>
            global_costmap:
                global_frame: map
                robot_base_frame: base_link
                update_frequency: 2.0  
                publish_frequency: 2.0
                static_map: true
                transform_tolerance: 2.0  
                clearing_rotation_allowed: false  
                width: 100.0
                height: 100.0
                origin_x: -50.0
                origin_y: -50.0
                resolution: 0.05
                plugins:
                  - {name: static_layer, type: "costmap_2d::StaticLayer"}
                  - {name: inflation_layer, type: "costmap_2d::InflationLayer"}
        </rosparam>
        
        
        <rosparam>
            local_costmap:
                global_frame: map
                robot_base_frame: base_link
                update_frequency: 10.0
                publish_frequency: 10.0
                static_map: false
                rolling_window: true
                width: 5.0
                height: 5.0
                transform_tolerance: 1.0
                plugins:
                  - {name: obstacle_layer, type: "costmap_2d::ObstacleLayer"}
                  - {name: inflation_layer, type: "costmap_2d::InflationLayer"}
        </rosparam>
    </node>

    
    <node name='neupan_control' pkg="neupan_ros" type="neupan_node.py" output="screen">
        <param name="config_file" value="$(arg config_file)"/>
        <param name="map_frame" value="$(arg map_frame)"/>
        <param name="base_frame" value="$(arg base_frame)"/>
        <param name="lidar_frame" value="$(arg lidar_frame)"/>
        <param name="scan_range" value="$(arg scan_range)"/>
        <param name="scan_angle_range" value="$(arg scan_angle_range)"/>
        <param name="marker_size" value="$(arg marker_size)"/>
        <param name="scan_downsample" value="$(arg scan_downsample)"/>
        <param name="dune_checkpoint" value="$(arg dune_checkpoint)"/>
        
        
        <param name="refresh_initial_path" value="true"/>
        
        
        <remap from="/scan" to="/limo/scan"/>
        
        <remap from="/neupan_cmd_vel" to="/cmd_vel"/>
        <remap from="/neupan_goal" to="/move_base_simple/goal"/>
    </node>

    
    <node name="rviz" pkg="rviz" type="rviz" args="-d $(find neupan_ros)/example/gazebo_limo/rviz/limo_gazebo.rviz" />
</launch>

PestoQing, 2025-07-27:
neupan_core.py的代码

#!/usr/bin/env python

"""
neupan_core is the main class for the neupan_ros package. It is used to run the NeuPAN algorithm in the ROS framework, which subscribes to the laser scan and localization information, and publishes the velocity command to the robot.

Developed by Ruihua Han
Copyright (c) 2025 Ruihua Han <hanrh@connect.hku.hk>

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
GNU General Public License for more details.

You should have received a copy of the GNU General Public License
along with this program. If not, see <https://www.gnu.org/licenses/>.
"""

from neupan.neupan import neupan
import rospy
from geometry_msgs.msg import Twist, PoseStamped, Quaternion, PoseWithCovarianceStamped
from nav_msgs.msg import Odometry, Path
from visualization_msgs.msg import MarkerArray, Marker
from sensor_msgs.msg import LaserScan, PointCloud2
from math import sin, cos, atan2
import numpy as np
from neupan.util import get_transform
import tf
import sensor_msgs.point_cloud2 as pc2

class neupan_core:
    def __init__(self) -> None:

        rospy.init_node("neupan_node", anonymous=True)

        # ros parameters
        self.planner_config_file = rospy.get_param("~config_file", None)
        self.map_frame = rospy.get_param("~map_frame", "map")
        self.base_frame = rospy.get_param("~base_frame", "base_link")
        self.lidar_frame = rospy.get_param("~lidar_frame", "laser_link")
        self.marker_size = float(rospy.get_param("~marker_size", "0.05"))
        self.marker_z = float(rospy.get_param("~marker_z", "1.0"))

        scan_angle_range_para = rospy.get_param("~scan_angle_range", "-3.14 3.14")
        self.scan_angle_range = np.fromstring(
            scan_angle_range_para, dtype=np.float32, sep=" "
        )

        self.scan_downsample = int(rospy.get_param("~scan_downsample", "1"))

        scan_range_para = rospy.get_param("~scan_range", "0.0, 5.0")
        self.scan_range = np.fromstring(scan_range_para, dtype=np.float32, sep=" ")

        self.dune_checkpoint = rospy.get_param("~dune_checkpoint", None)
        self.refresh_initial_path = rospy.get_param("~refresh_initial_path", True)
        self.flip_angle = rospy.get_param("~flip_angle", False)
        self.include_initial_path_direction = rospy.get_param("~include_initial_path_direction", False)

        if self.planner_config_file is None:
            raise ValueError(
                "No planner config file provided! Please set the parameter ~config_file"
            )

        pan = {'dune_checkpoint': self.dune_checkpoint}
[... 475 lines cut ...]
        return marker

    @staticmethod
    def yaw_to_quat(yaw):

        quater = Quaternion()

        quater.x = 0
        quater.y = 0
        quater.z = sin(yaw / 2)
        quater.w = cos(yaw / 2)

        return quater

    @staticmethod
    def quat_to_yaw(quater):

        x = quater.x
        y = quater.y
        z = quater.z
        w = quater.w

        raw = atan2(2 * (w * z + x * y), 1 - 2 * (pow(z, 2) + pow(y, 2)))

        return raw